eldruin/pwm-pca9685-rs #3: Updating using fewer writes

mrd0ll4r, 2019-11-21:
Hey!

We're working on some cool home lighting stuff using, among others, multiple PCA9685s for a ton of LED strips. We're looking to reduce the number of syscalls and the overhead on the I2C bus due to small transactions, because we might actually be hitting some limits there :sweat_smile: 

So, I was wondering: Instead of setting ON and OFF for each channel separately, it seems to be more efficient to just set them at the same time in order to save some bandwidth? I came up with something like this:

```rust
    fn write_quad_register(
        &mut self,
        address: u8,
        value0: u16,
        value1: u16,
    ) -> Result<(), Error<E>> {
        if self.config.is_low(BitFlagMode1::AutoInc) {
            let config = self.config;
            self.write_mode1(config.with_high(BitFlagMode1::AutoInc))?;
        }
        self.i2c
            .write(
                self.address,
                &[
                    address,
                    value0 as u8,
                    (value0 >> 8) as u8,
                    value1 as u8,
                    (value1 >> 8) as u8,
                ],
            )
            .map_err(Error::I2C)
    }
```

However, I'm not too proficient in Rust (especially with macros) and don't know how to nicely integrate this with your `impl_channel_match` macro. I came up with this:

```rust
macro_rules! impl_channel_match_quad {
    ($s:ident, $channel:expr, $value0:expr, $value1:expr, $($C:ident, $reg:ident),*) => {
        match $channel {
            $(
                Channel::$C  => $s.write_quad_register(Register::$reg, $value0, $value1),
            )*
        }
    };
}

impl<I2C, E> Pca9685<I2C>
where
    I2C: hal::blocking::i2c::Write<Error = E>,
{
...
    /// Set the `ON` and `OFF` counters for the selected channel.
    /// 
    /// Note that the full off setting takes precedence over the `on` settings.
    /// See section 7.3.3 "LED output and PWM control" of the datasheet for
    /// further details.
    pub fn set_channel_on_off(
        &mut self,
        channel: Channel,
        on: u16,
        off: u16,
    ) -> Result<(), Error<E>> {
        if on > 4095 || off > 4095 {
            return Err(Error::InvalidInputData);
        }
        impl_channel_match_quad!(
            self, channel, on, off, C0, C0_ON_L, C1, C1_ON_L, C2, C2_ON_L, C3, C3_ON_L, C4,
            C4_ON_L, C5, C5_ON_L, C6, C6_ON_L, C7, C7_ON_L, C8, C8_ON_L, C9, C9_ON_L, C10,
            C10_ON_L, C11, C11_ON_L, C12, C12_ON_L, C13, C13_ON_L, C14, C14_ON_L, C15, C15_ON_L,
            All, ALL_C_ON_L
        )
    }
...
}
```

I didn't open a PR yet because, as mentioned, I'm not too comfortable with Rust yet. But I'd be happy to hear your feedback and then open one!

Alternatively, or taking this one step further: Would it be possible to write all of the 64 LED output registers in one transaction? Reading through the Linux [i2cdev docs](https://www.kernel.org/doc/Documentation/i2c/dev-interface) gives me the feeling that 32 bytes might be the limit for one transaction (at least when using SMBus?), so this might actually become two writes instead of one.

Please excuse if I'm talking nonsense - I don't do much embedded programming, so I am a bit confused...

eldruin, 2019-11-21:
Hi @mrd0ll4r,
This is interesting. I think I can achieve everything you proposed.
I would not think that the Linux I2C impl would split simple write array transactions ([we are not using SMBus transactions](https://github.com/rust-embedded/rust-i2cdev/blob/c2611e233b45f0339bde63debd2feb6e759fe779/src/linux.rs#L136)).
I will implement something in the next couple of days and let you know here.

eldruin, 2019-11-21:
Ok @mrd0ll4r, I added the functions `set_channel_on_off()` and `set_all_on_off()`.
Do they fit your use case?

mrd0ll4r, 2019-11-22:
Amazing! Yes, that's exactly what I was looking for, thank you so much!
The parts have not arrived yet, so I can't test it unfortunately, but I don't see why it wouldn't work.
Thank you! :)

eldruin, 2019-11-22:
Great, then I will move ahead with a patch release.
It would be great if you would share a video when done :)

eldruin, 2019-11-22:
Published in version [0.1.2](https://github.com/eldruin/pwm-pca9685-rs/releases/tag/v0.1.2)

closed 2019-11-22 by eldruin

mrd0ll4r, 2020-01-27:
Hey! Just wanted to let you know that this works like a charm!
I don't have a video (yet), because we're still building the stuff around it all, but we prototyped a bit and looked at the bus with a logic analyzer, and I can confirm that all values are written in one block and it's nice and speedy!

Thank you :)

eldruin, 2020-01-27:
I am very happy to hear that. Thanks for letting me know :)