eldruin/pwm-pca9685-rs #7: can't get the library to work

LinusDikomey, 2021-03-28:
Hey,
I'm trying to control a servo and some LEDs with the library but I can't get it to output anything at all. To confirm that everything is connected correctly I used a simple python script doing exactly the same and it worked. What could I be doing wrong?

Rust program that doesn't work:
```Rust
use linux_embedded_hal::I2cdev;
use pwm_pca9685::{Address, Channel, Pca9685};
use std::{thread, time::Duration};

fn main() {
    let dev = I2cdev::new("/dev/i2c-1").unwrap();
    let address = Address::default();
    let mut pwm = Pca9685::new(dev, address).unwrap();

    // This corresponds to a frequency of 60 Hz.
    pwm.set_prescale(100).unwrap();

    loop {
        println!("Setting to 150");
        pwm.set_channel_on_off(Channel::C0, 0, 150).unwrap();
        thread::sleep(Duration::from_secs(3));
        println!("Setting to 600");
        pwm.set_channel_on_off(Channel::C0, 0, 600).unwrap();
        thread::sleep(Duration::from_secs(3));
    }
}
```

Python script that works:
```python
from __future__ import division
import time

import Adafruit_PCA9685

pwm = Adafruit_PCA9685.PCA9685()

servo_min = 150
servo_max = 600

pwm.set_pwm_freq(60)

print('Moving servo on channel 0, press Ctrl-C to quit...')
while True:
    print(servo_min)
    pwm.set_pwm(0, 0, servo_min)
    time.sleep(3)
    print(servo_max)
    pwm.set_pwm(0, 0, servo_max)
    time.sleep(3)

```

eldruin, 2021-03-28:
Hi, could you add an enable with `pwm.enable().unwrap();` and try again?

LinusDikomey, 2021-03-28:
> Hi, could you add an enable with `pwm.enable().unwrap();` and try again?

Thank you! That was the problem.

closed 2021-03-28 by LinusDikomey