eldruin/pwm-pca9685-rs #15: Update to embedded-hal 1.0

bollian, 2024-01-31:
Are there any plans to update to the new stable release of embedded-hal 1.0? It might be a little tricky since a couple big changes have been made since 0.2:
- all blocking I2C operations are grouped together under a single trait: `embedded_hal::i2c::I2c`
- the I2C trait now supports both 7bit and 10bit addresses using a generic address parameter

Thankfully, the actual `write` and `write_read` methods don't seem to have changed at all, aside from the address size.

eldruin, 2024-01-31:
Upgrading to embedded-hal 1.0 should be very simple here. Just changing the imports and grouping the write and write_read operations.
However, I wanted to give some time to @nils-van-zuijlen who is busy working on async support.

nils-van-zuijlen, 2024-01-31:
I am currently waiting on an answer in https://github.com/nvksv/maybe-async-cfg/issues/2, and upgrading to have only a single I2C trait may make it easier to toggle between them.

eldruin, 2024-01-31:
@nils-van-zuijlen Last week @andresv used `maybe-async-cfg` to add async support to the tcs3472 driver in [this PR](https://github.com/eldruin/tcs3472-rs/pull/5). Maybe that is helpful to you.

PR #16 (2024-02-02): Use maybe-async-cfg to add async support
Superseeds #14 , #13 

Also, update to embedded-hal v1 so that there is only one trait for I2C on both sides.

Closes #15 
Closes #11 

Heavily inspired by https://github.com/eldruin/tcs3472-rs/pull/5

I could not make the example in `lib.rs` compile, even when adding the embassy dependencies, because I cannot figure out how to enable a feature for a single doctest. So I did not include the embassy deps.
closed 2024-02-02 by eldruin